krontech/chronos-cam-app #35: Stability issues after large numbers of saves

skronstein, 2019-03-01:
After saving hundreds or thousands of videos, the camera may lock up on the playback/save window.

oskirby, 2019-03-17:
The chronos-cli repository has seen some work lately that should improve the stability as best as we can get it from the Cortex-A8 side of things. However, the firmware for the video core has still exhibited the occasional crash that we have yet to track down. MTBF seems to be somewhere in the 2-5k recordings at present.

See krontech/chronos-cli@d15a55872f1ab9088a161c3031a62cc4b92c1d87 for more info.

oskirby, 2020-03-04:
The issue was ultimately traced down to an old firmware version that was being loaded into the VPDMA accelerator in the dm8148. The loading of this firmware is initially done by u-boot to bring up the boot splash, but due to a bug in the CPU the code that would attempt to reload the VPDMA firmware during the boot of the M3 cores would leave the old firmware in place.

This was resolved by updating the VPDMA firmware that gets loaded by u-boot to match what the M3 cores are expecting. And the fix was deployed as part of the u-boot-ti81xx Debian package (closest commit would be here: https://github.com/krontech/chronos-updates/commit/53937dc9fb74ce79d02f928bcc1d1459fc5d00f1)