krontech/chronos-cam-app #36: Some settings not restored after reboot

skronstein, 2019-03-06:
The following settings are sometimes set to off after a reboot, even if they were set to on before the reboot:
UI on Left Side
Focus Peaking
Enable Zebras

skronstein, 2019-03-15:
I can't find a way to restore focus aid and zebras correctly on reboot.

I tried calling Camera::setFocusPeakEnable() at different locations in CamMainWindow's constructor, but none worked consistently - before or after gpmc->init(), or at the end of CamMainWindow's constructor.

I also tried using QTimer::singleShot to wait a second or more before calling on_chkFocusAid_clicked().

At least it changes when toggling focus aid or coming back from Util.

oskirby, 2019-03-25:
The recent commit https://github.com/krontech/chronos-cam-app/commit/3c7b2bea43bf59d0f1fc22fdcbc28213ef782baa should have resolved this issue by moving the entirety of focus peaking and zebra management into the video pipeline daemon, however, I still have yet to reproduce. I'm going to close this until such time as we can come up with some steps to reproduce the issue that you are seeing.

closed 2019-03-25 by oskirby