krontech/chronos-cam-app #38: White balance has opposite effect

skronstein, 2019-03-08:
Setting custom white balance in warm or cool light just makes the image even warmer or cooler.
Presets also have the opposite effect intended.

skronstein, 2019-03-12:
Fixed  https://github.com/krontech/chronos-cam-app/commit/353ed31866ba2a140f7948499e103852da312ad4

oskirby, 2019-03-17:
This change disagrees with the FPGA register mapping, I suspect something else is the cause of your color issues here.

oskirby, 2019-03-18:
After some further testing, this looks like a bug in FPGA version 3.15 and newer builds of the FPGA have the RGB channels in the correct order when applying the white balance. I think we need to revert this change and update the FPGA image instead for the 0.3.1 release.

closed 2019-03-18 by oskirby