krontech/chronos-cam-app #39: Record Settings -

skronstein, 2019-03-09:
After using the keyboard to enter a horizontal resolution below 192, or a vertical resolution below 96, the vertical or horizontal resolution just reverts to what it was before the keyboard was opened, instead of going to 192 or 96.

oskirby, 2019-03-15:
When an invalid frame size is entered the UI should attempt to find the nearest valid setting instead of reverting to the previous valid. (eg: values below 192 are set to 192 instead).

skronstein, 2019-03-16:
Fixed.
https://github.com/krontech/chronos-cam-app/commit/a736c8a8ab738017795c291e9ad684f7fc6eaf49

skronstein, 2019-07-31:
H resolution will still revert to previous instead of minimum if you enter a resolution below min, then tap the V res text.  But this only happens to the horizontal resolution spinbox, not the vertical resolution one.

skronstein, 2019-09-19:
Fixed again: https://github.com/krontech/chronos-cam-app/commit/a964c7240735a2af0674c8c669fc27b9ad457611

closed 2019-09-19 by skronstein