krontech/chronos-cam-app #43: Trigger Input 3 - Invert doesn't have an influence on the recording trigger

kbmag126, 2019-03-12:
camUpdate-v0.3.1-rc1-4-g112da3a

Enabling invert doesn't seem to have an influence on the trigger. Trigger 3 will always activate once the signal reaches 1.5V regardless if a rising or falling signal is applied. This behavior is different to how IO1 and IO2 reacts. 
This behavior is also different to what was mentioned in the manual. Once 1.5V is reached the camera will recognize it as "HIGH", but this should only be "HIGH" when a 3 - 30V signal is applied.

oskirby, 2019-03-15:
We should look into if the FPGA is being set correctly to enable trigger 3 invert, it may be that the UI is simply configuring the FPGA incorrectly.

oskirby, 2019-03-20:
This appears to be resolved in FPGA build 3.21 added in https://github.com/krontech/chronos-updates/commit/8e4458e75ad4f4c9cf975d06c29eb52957ce7c72

closed 2019-03-20 by oskirby