krontech/chronos-cam-app #44: Record Mode - Gated Burst Error

kbmag126, 2019-03-12:
camUpdate-v0.3.1-rc1-4-g112da3a

Occurs to all Trigger IO for the test
This was tested on 2 different cameras, 00020 (color 16GB) and 00031(color 16GB)

Performing the gated burst test resulted in recordings with 2 frames or recordings with high number of frames(70,000 - 400,000). The recording (3 seconds) would usually be around 3000 frames. 

Video recording is damaged. Various vertical lines patterns shown in the video playback. Vertical lines on video playback is changing on its own. May be accessing previous recording data on the RAM since the video is showing some recordings at completely different times.

Reverting to defaults may allow you to get a correct recording, but subsequent recordings will have the errors mentioned above

oskirby, 2019-03-15:
We would like to see this test repeated with debouncing enabled on the triggers to be sure that button bounce isn't contributing to the issue.

However, in any case there should not be any garbled frames in memory.

kbmag126, 2019-03-18:
Tested on 00020 (16GB) and 01006 (16GB)
Problem still occurs on all Trigger IOs. 
Still getting recordings that is either 2 frames or a high number of frame recording. Enabling Debounce did have a small influence on the number of frames in the recording. High number of frame recordings is now between 70000 to 140000.
Garbled video recordings mentioned in the first post still occurs.

oskirby, 2019-03-20:
This might be resolved by the segmented video fix in https://github.com/krontech/chronos-updates/commit/1482f8005ca9a0cc14d764a1c663d438cb57267a

kbmag126, 2019-03-22:
Tested on 00020 and 01006
Occurs for TRIG1, TRIG2, and TRIG3

No longer having high frame recordings, but there are still problems. Still generating recordings that are 2 frames or recordings that are around 8000 frames. Still jumping to recordings of different time periods in the video recording. Occasionally getting recordings with the incorrect number of frames that does have a 3 second recording. However, it is producing videos with 500 - 800 frames instead of 3000 frames. 
Can get a correct recording of 3000 frames after turning the camera on and off or reverting to defaults, but subsequent recordings still producing problem recordings.

oskirby, 2019-03-24:
I have been able to reproduce the issue at least. At first glance I think this is an FPGA issue. After instrumenting the cam-pipeline to trace out all the segments that are generated when in gated burst I find that the second recording attempt generates mostly identical segment information to the first, but with the occasional group of insane values throw in. If the insane segment happens to overlap with a valid one then it results in a loss recording segments - giving rise to the sub-3000 frame recordings that you have observed.

oskirby, 2019-03-24:
I stand corrected, the issue is the generation of the D-Bus segment signal taking up too much time, which blocks the thread from being able to service the recording FIFO. This bug would have been introduced in https://github.com/krontech/chronos-cli/commit/a4d0c9075f3acf1295398cfb993f5d1fe12fa4e5

So, another approach will be needed to generate (and probably rate-limit) the segment information over D-Bus when operating in gated burst mode.

closed 2019-03-24 by oskirby