krontech/chronos-cam-app #49: Gated burst - Live display is not buffered while trigger inactive.

oskirby, 2019-03-24:
When the camera is operated in gated burst mode, the live display does not appear to be buffered correctly when the trigger signal is inactive.

To reproduce this issue, set up the camera to film a fast-moving and repetitive event (a spinning fan makes an excellent subject), configure the camera for gated burst mode, and then start recording. While the trigger signal is inactive the video appears to be torn at multiple horizontal positions across the fan. The frame tearing appears to go away when the trigger is activated.

Tested on FPGA build 3.21.

oskirby, 2019-03-24:
[image]

tesla5hundred, 2019-06-14:
This is expected behavior. When trigger is inactive, the live display is pulled from the single frame that is being continuously overwitten multiple times per display frame.