krontech/chronos-cam-app #65: Standard black cal makes the image look very dark at low framerates (below about 45fps)

skronstein, 2019-06-22:
At 30 and 45fps, the image looks OK until standard black cal is performed, then the image looks very dark.
At 60fps, the image looks OK both before and after standard black cal.
tested on latest 3.2 beta

oskirby, 2020-03-04:
This can be addressed by adding an extra PRSTN pulse before the start of exposure, the difficulty is that said PRSTN pulse cannot occur while the wavetable is active, so we wind up needing a different timing program depending on whether the exposure overlaps with the wavetable or not.

The 'Global Control Pulses' section of the LUX1310 datasheet describes how the PRSTN signal can be used to correct for the dark current that accumulates in the storage nodes during very long frame periods.

oskirby, 2020-03-04:
This is likely to be impossible to resolve in shutter gating mode, since the timing program can't dynamically determine when to insert this control pulse or not. For normal operation and frame trigger modes, this extra control pulse could be inserted depending on the frame period.