krontech/chronos-cam-app #70: False touch events generated in high wind.

oskirby, 2019-07-07:
When exposed to a high-end environment, it has been observed that false touch events can be generated on the LCD screen. This makes the camera difficult if not impossible to use in some environments.

The incidence of false events has been reported to depend on the strength and angle of attack of the wind.

skronstein, 2019-07-08:
Doesn't smoke worsen this too?

kaklik, 2021-07-10:
> When exposed to a high-end environment, it has been observed that false touch events can be generated on the LCD screen. This makes the camera difficult if not impossible to use in some [...]

Did you remove the covering foil from the LCD?  We use the camera in relatively harsh environments (thunderstorms) and we, therefore, have the original covering foil in place.  I do not notice false touch events until now.