krontech/chronos-cam-app #74: Max Bitrate adjusts max megabits per 60 frames instead of the actual max bitrate

skronstein, 2019-07-19:
If the saved file frame rate is set lower than 60fps, the maximum bitrate will not adjust the max bitrate, but instead the max megabits per 60 frames.

skronstein, 2019-07-19:
Perhaps either the UI should indicate that this adjusts the max megabits per 60 frames, or the maximum settable UI value for Max Bitrate should be adjusted each time Saved File Framerate is adjusted.
The user manual already somewhat addresses this, as it says the following: "If the saved file frame rate is set lower than 60fps, the maximum bitrate will be lower than 60Mbps."

oskirby, 2019-07-24:
I think 'maximum megabits per 60 frames' is not a metric that makes a lot of sense and we should fix the UI to display megabits per second correctly.

skronstein, 2020-01-21:
Fixed: https://github.com/krontech/chronos-cam-app/commit/7fb947e598e40d6548b09cc07fb97e5473ae0ad5

closed 2020-01-21 by skronstein