krontech/chronos-cam-app #86: Vertical lines in image after Auto Factory Cal

skronstein, 2019-08-21:
After performing an Auto Factory Cal, there are vertical lines in the image.  These remain until a black cal is performed.

oskirby, 2019-09-26:
This appears to be a result of the ADC offset being inconsistent with resolution changes. In particular, generating an ADC offset calibration at 1280x1024 yields a different result from generating an ADC offset calibration at, say, 1280x720. Since the ADC offset is only stored once per gain/wavetable combination this means that the two ADC offset calibrations will overwrite each other.

This can also be reproduced by doing a black cal at one resolution and switching to another resolution that would use the same offset file. The variation in ADC offsets without re-calculating FPN results in the vertical banding seen in the image.

oskirby, 2019-09-26:
There are two approaches that we could take to solve this:

1. Store the ADC offset training data uniquely per resolution.
2. Remove ADC offset training from being done on each black cal, so that the ADC offsets remain consistent.

closed 2019-10-08 by oskirby