krontech/chronos-cam-app #91: Util screen - sometimes "clicking" the tabs does not change to them but only highlights them

skronstein, 2019-09-05:
When one of the new LCD modules is installed, and a tab is tapped in the Util menu, that tab is sometimes only highlighted, not switched to.
Reported by NiNeff on the forum, who thinks it only started happening after upgrading his LCD: http://forum.krontech.ca/index.php?topic=488.msg2705#msg2705
I haven't been able to reproduce it on my camera, which has an old LCD module.

skronstein, 2019-09-05:
I will try this with a camera that has a new module soon.

oskirby, 2019-09-24:
I think we are confounding two issues here:

The first issue that the title of the bug report is describing is that there are a few places on the screen that, when touched, can change UI focus without clicking the widget. This has always been the case with the Qt GUI, it's just that we don't usually notice it. This probably isn't worth changing.

The second issue is that the response of the touch screen may not be in perfect alignment with the LCD display. Normally this would be fixed using calibration software to align the touch inputs with the underlying LCD. On the old panel, the geometry was close enough that we have never really needed to perform that step, we may need to update the tslib calibration data for the new panel.