krontech/chronos-cam-app #92: Can't format a USB device if it is mounted as sdb instead of sda

skronstein, 2019-09-13:
Also, it doesn't show under USB/SATA storage, but it does under Mounted Devices. See the screenshot.
[image]

oskirby, 2019-09-24:
How exactly do you get a USB drive to enumerate as /dev/sdb? That shouldn't be possible unless something has gone wrong.

Do you have any steps to reproduce this?

skronstein, 2019-09-25:
I'm not sure why it is sometimes mounted as sdb. I just remember that, sometimes when I insert a flash drive after booting, it gets mounted as sdb.
LoialOtter thought this might happen if the user removes and quickly reinserts a storage device.  I tried removing and reinserting it about 10 times and it showed up as sda every time.

Knackie, 2019-10-09:
try with Gparted