krontech/chronos-cam-app #93: Increase in memory not detected in playback screen

sanjpm, 2019-09-13:
Upgrading memory in the camera from 8 GB to 16GB, or 16 to 32 will correctly show the new space in the About screen, but the end save frame value in the playback window stays at 8731. 

Upgrading the camera to the same version it is currently on does not fix the issue, but downgrading and then upgrading seems to fix it. Restarting camApp via qtcreator will also fix it temporarily.

First appearance on 0.3.1 as reported by production.

skronstein, 2019-09-16:
After going to Record Settings->Record Modes and changing the length of the recording there, it should then be possible to make a longer recording.
Ideally, when the camera detects a change in memory, it should scale the recording length to the new maximum.  Users don't upgrade their camera's memory very often, so this is indeed a minor issue.

skronstein, 2019-09-17:
After going to Record Settings->Record Modes and changing the length of the recording there, it should then be possible to make a longer recording.
Ideally, when the camera detects a change in memory, it should scale the recording length to the new maximum.  Users don't upgrade their camera's memory very often, so this is indeed a minor issue.

The camera already has a function to detect the amount of memory installed, so I could just use that to program in some functionality to fix this. It would compare the memory currently installed to that which was installed on the previous boot. If it's different, it will update the recording length to the new maximum.