krontech/chronos-cam-app #97: Auto Power Mode always shows Disabled regardless of what is set

skronstein, 2019-10-22:


skronstein, 2019-11-01:
This bug happens only occurs on software v0.3.2-RC2, not RC1. The version of pcUtil shipped in RC2 is what's causing that bug with the auto power setting.

In this commit, it was updated to the version that has the issue, and that binary is only about 60kb in size, instead of the 300kb for the one in RC1. https://github.com/krontech/chronos-updates/commit/bf78ada8d858467e6160c5727fcaa198977260bf

Sanjay said it was built with this version of pcUtil: https://github.com/krontech/chronos-cli/commit/1e3868b4250bb3159d6353863890417b35b3f0c3#diff-abb86635b24009f3799d0303735dc658

oskirby, 2019-11-02:
This appears to be due to a disconnect between camApp and pcUtil regarding who is responsible for triggering battery data updates. camApp is polling for battery data at the same time as pcUtil is automatically generating battery data at approximately 1Hz. This means that battery data reports are backing up in the communication socket, and when the camera tries to set one of the power modes, it gets a battery data instead of the expected "success" message and assumes that the operation failed.

This had already been fixed in the master branch, and was resolved by treating all responses from pcUtil as asynchronous messages.

See: https://github.com/krontech/chronos-cam-app/commit/2ccb45cfe8807e6f2a2012e5d45f81a97a0940b2

closed 2019-11-02 by oskirby