krontech/chronos-cam-app #99: Overlay shows incorrect time if recording was stopped early while trigger delay enabled

skronstein, 2019-10-29:
Steps to reproduce:
1. Set a trigger delay - 0% makes this "bug" most obvious
2. Start a recording, and stop it early at, for example, 1 second after starting.
3. Enter the playback window and check the time reported by the text overlay.  When it calculates the time, it seems to calculate it assuming that the trigger happened [number of post-trigger frames] frames before the end of the recording, even if the recording is shorter than that, and even if the recording was stopped using the shutter button instead of the trigger.

oskirby, 2019-11-13:
This is probably due to the difference between a 'trigger' and the stop recording button. Triggers go through a bunch of timing blocks in the FPGA before their effect on the recording program happens while the stop recording buttons will always terminate the recording immediately regardless of whatever delay is configured.

However, when the video segment data is generated by the FPGA there is no way for the video system to know under which conditions the recording was terminated, and therefore it always assumes it was terminated by the trigger delay mechanism (and therefore uses the trigger delay settings in computing frame timestamps).

oskirby, 2019-11-13:
To solve this problem, the FPGA would need to be modified to include the 'reasons why the segment ended' to the video segment data as well.