krontech/chronos-cam-app #103: At some exposure settings, black cal doesn’t work 100% correctly

skronstein, 2019-12-04:
At some exposure settings, black cal doesn’t work 100% correctly or takes multiple tries to do so. I saw the following anomalies at 1280x720 max fps:

At 294us exposure, the image showed a horizontal bar that disappeared after a second black cal.

At 96.86us exposure, the image always showed a thin horizontal bar near the bottom, even after multiple extra black cals.

oskirby, 2020-03-04:
I think this issue is a duplicate of https://github.com/krontech/chronos-cam-app/issues/149

closed 2020-03-04 by oskirby

skronstein, 2020-03-06:
> I think this issue is a duplicate of #149

This happened on the 1.4, so I don't think it would be.  If i'm wrong, you can leave it closed.