krontech/chronos-cam-app #107: Text overlay always assumes frame period of 4ms

skronstein, 2019-12-07:
When the overlay is enabled and I play a recording, the time as reported by the overlay advances by .004s regardless of what framerate the video was recorded at.

skronstein, 2019-12-07:
Also:
With no pre-record delay, the time at the end of the recording is shown correctly (0 seconds).
But with a pre-record delay, the time shown is incorrect.

oskirby, 2020-01-07:
The code that calculates the frame period for the text overlay was written using the old timing block. Unfortunately, this means that when the new timing block is being used the pipeline has no way to easily determine the frame period or exposure time when the new timing programs are in use.

I think the best way to fix this would be to add a register or two to the FPGA that count the current frame period and exposure. This would also possibly enable the overlay to determine the period and exposure timing when external shutter of frame sync is used.

closed 2020-03-05 by skronstein