krontech/chronos-cam-app #112: Auto power mode - camera doesn't Turn on if AC inserted

skronstein, 2019-12-11:
Even if Auto Power mode is set to "Turn off if AC removed" or "Both," it doesn't turn off when AC is removed.

sanjpm, 2019-12-11:
Output from journalctl:

Dec 10 19:34:48 chronos camApp[321]: "Failed to set parameter powerOffWhenMainsDisconnected: Unknown property 'powerOffWhenMainsDisconnected' could not be set."

skronstein, 2020-01-09:
Possibly related:
Auto Power Mode intermittently resets to Turn On if AC Inserted https://github.com/krontech/chronos-cam-app/issues/137

oskirby, 2020-01-11:
This may also need a fix to cam-pcUtil to correctly read the power modes back from the PMIC or you may experience some UI inconsistency in the state of the auto-power mode combo box after a reboot.

https://github.com/krontech/chronos-cli/commit/bf0fc10326eaef93590611aeeb139f8d74dbc2ff

closed 2020-01-11 by oskirby