krontech/chronos-cam-app #113: Record Settings - Max exposure button doesnt set correct max if framerate is below maximum

skronstein, 2019-12-11:
The 'Max' button beside Exposure always sets the exposure based on max framerate for current resolution, not current framerate.

oskirby, 2020-01-08:
The exposure limits returned by the API are slightly incorrect, and will return the maximum exposure at the minimum frame period. Where the UI really needs to know the maximum exposure for a given frame period.

The v0.3.2 software simply assumed that the maxExposure = framePeriod - 500ns. We should probably provide this 500ns somewhere in the API so that the UI can calculate the value it needs. This should probably be done by adding another field to the getResolutionTimingLimits method with this overhead timing (eg: 500ns on the LUX1310)

oskirby, 2020-01-10:
There is a slightly easier solution, since the getResolutionTimingLimits method accepts a frame geometry structure (which includes a frame time), we can simply make it return the exposure range for that frame time.

This kinda muddies the waters a bit with this API call since the question of 'minimum frame period' might still not be 'the current frame period' but for the purposes of camApp it does the job with minimal API changes.

closed 2020-01-10 by oskirby