krontech/chronos-cam-app #122: Debounce can't be enabled

skronstein, 2019-12-12:
After trying to enable Debounce, it seems to not have any effect, at least for IO 1.  Haven't tested the other 2.
For all 3 IOs, it always shows up as Disabled after returning to the IO screen even if it was checked and Apply was pressed before pressing OK.

skronstein, 2020-01-06:
Debounce now works for me on input 2, but still not for inputs 1 or 3.

oskirby, 2020-01-06:
I probably broke something. My apologize for being a garbage developer.

oskirby, 2020-01-06:
After a bit of testing, it seems that the behaviour can more accurately be described as the state of the debounce checkbox is not saved when there is no input function selected. Which in a way kindof makes sense because there is "nothing to debounce" in this case.

However, for user sanity, we probably should retain some way to keep the state of the checkbox even when there isn't actually an API thing that is controlling in this case.

oskirby, 2020-01-06:
This also probably affects the invert button as well for the same reason.

closed 2020-01-10 by oskirby