krontech/chronos-cam-app #129: incorrect default color matrix

skronstein, 2019-12-19:


oskirby, 2020-01-06:
Can you clarify what the expected behavior is supposed to be here? Hitting the default button on the color matrix screen appears to load the coefficients for the CIECAM16/D55 matrix.

skronstein, 2020-01-07:
Sorry, I didn't bother writing a description because I was going to look into it myself, but I probably should have clarified what I meant by default anyway.

Steps to reproduce the bug:
1. Revert settings by deleting /var/camera/Settings/KronTech/camApp.conf
2. Reboot the camera
3. Open the white balance -> Matrix window and check which values show up in the spinBoxes on the Matrix window.  They will be the CIECAM02 values instead of the CIECAM16 values.

oskirby, 2020-01-08:
I think the real problem here is that the camera settings are no longer retained entirely in Qt and are instead spread across Qt, pychronos and cam-pipeline. We need to add an API call to revert and backup the camera config.

skronstein, 2020-01-21:
Closing this, since it is basically a duplicate of: Revert/restore defaults not working https://github.com/krontech/chronos-cam-app/issues/101

closed 2020-01-21 by skronstein