krontech/chronos-cam-app #132: IO - Frame Sync Output has no effect without pullup

skronstein, 2020-01-03:
At least for IO 1, the camera does not output anything when frame sync output is enabled, unless a pullup is enabled.

oskirby, 2020-01-06:
Isn't this how the camera has always operated? The output IO structure of triggers 1 and 2 are just current pull-ups and if all the pullups are disabled then the camera isn't physically capable of outputting anything.

skronstein, 2020-01-07:
On 3.2, frame sync output works even without any pullup enabled, and the user experience should stay the same.

oskirby, 2020-01-11:
On the old IO block, the pullups are implicitly enabled whenever the frame sync output is selected. However, the behavior of the new timing block is a fairly good representation of how the hardware works and it would be weird to add a special case for the frame-sync mode, so I think that the cleanest way to approach this problem is to keep the API the same, but to make the UI enable the pullups when frame-sync output is desired.

closed 2020-01-11 by oskirby