krontech/chronos-cam-app #138: White balance window - Custom white balance added multiple times, and freeze

skronstein, 2020-01-09:
While testing the white balance window on software V0.3.1 last year, I tapped Set Custom 6 times, and with each press, there was a new Custom entry in the list of white balances available under the White Balance dropdown. After I pressed it a 7th time, camApp crashed.

I've been unable to reproduce these issues.  I looked at the source code and didn't find anything that would either allow multiple Custom entries in the dropdown, or would crash camApp.

oskirby, 2020-01-09:
This code has been rewritten a couple of times to work with the API instead of directly with the FPGA registers. If you are not able to reproduce this issue on the latest stable release (v0.3.2) or the API development branch (v0.4.0) then would consider the problem resolved.

skronstein, 2020-01-10:
Even on 3.1, I was unable to produce it a second time.

oskirby, 2020-01-18:
Given that we are not able to reproduce this issue, I am closing it.

closed 2020-01-18 by oskirby