krontech/chronos-cam-app #139: Corruption in segmented mode

skronstein, 2020-01-11:
I tested segmented record mode twice at 800x600 resolution and found that the 7th segment was corrupted both times. It The bug did not occur when I played back a recording with only 4 segments and same length per segment.
[image]

oskirby, 2020-01-21:
I think I need a bit more detail to figure out how to reproduce this issue. Can you tell me how the recording modes were configured and how exactly the test was run? (Eg: configured 8 segments of 1234 frames, then hit the trigger 7 times, and exited recording mode using the red button).

Also, how much memory does your camera have?

skronstein, 2020-03-06:
I am unable to reproduce this, at least on the latest software, so I'm closing this now.

closed 2020-03-06 by skronstein