krontech/chronos-cam-app #141: White balance - green is about 3% stronger than red/blue

skronstein, 2020-01-15:
After saving a video of a colorchecker in H.264 format, taking a screenshot from the video and examining the image in the GIMP, the white square does not appear exactly white, having the green channel about 3-4% stronger than red and blue.

skronstein, 2020-01-15:
Will retest on TIFF since that is now fixed.

oskirby, 2020-01-21:
I would definitely recommend using a raw format to test white balance since h.264 encoding doesn't do an awesome job of preserving chroma data after encoding.

It would also be good to know what the color maxtrix is for this test it is not necessarily guaranteed for the color matrix to preserve the white point. Retesting this with an identity color matrix *should* return exactly white.

skronstein, 2020-01-21:
I did all tests (both before and today) with an identity matrix.
TIFF, blue was 5% lower than the other channels.
All tests up to this point I had done at around 50% brightness (brightness values of the spot on which white balance was performed were around 128 according to GIMP), so I opened the aperture some more and tried again with both TIFF and H.264.  TIFF had all channels within 0.5% of each other, and H.264 within 1.5%. (brightness values around 128 according to GIMP) So, white balance appears to not work correctly at lower brightness for some reason.