krontech/chronos-cam-app #143: Zebra lines does not work in latest software

Alexy899, 2020-01-20:
Customer complained that when enabling zebra lines‘ is selected, the zebra lines don’t seem to work. 

Be able to reproduce this zebra issue in V0.3.2-patch 1, while V0.3.2 and V0.3.1 tested ok in same camera. 

is it possible to provide estimated time to solve this bug to customer?

Thanks

Alex

oskirby, 2020-01-20:
It looks like in between v0.3.2 and v0.3.2-patch1 a changed was added for the API development branch that got applied to the video system. Unfortunately it looks like this broke the use of zebra stripes in the version of cam-pipeline that was built for Arago cameras.

https://github.com/krontech/chronos-cli/commit/37ed8fa93740b6940d2579bf12e523157c3d1624

Alexy899, 2020-01-20:
Thanks Foobar for your prompt response.  Sent from Mail for Windows 10 From: oskirbySent: Monday, January 20, 2020 1:26 PMTo: krontech/chronos-cam-appCc: Alexy899; AuthorSubject: Re: [krontech/chronos-cam-app] Zebra lines does not work in latest software (#143) It looks like in between v0.3.2 and v0.3.2-patch1 a changed was added for the API development branch that got applied to the video system. Unfortunately it looks like this broke the use of zebra stripes in the version of cam-pipeline that was built for Arago cameras.krontech/chronos-cli@37ed8fa

closed 2020-01-20 by oskirby