krontech/chronos-cam-app #164: Software update: Selecting headless mode has no way out

oskirby, 2020-03-12:
When switching between GUI modes, it is possible that a user might select headless mode by accident, in which case there is no longer any GUI that a user might be able to use switch back to a "headed" mode.

This can be recovered by using SSH to log into the camera and re-enabling one of the GUI services (eg: `systemctl enable chronos-gui`)

oskirby, 2020-03-12:
One other possibility, is that in the event of some kind of catastraufic software problem, the selected GUI might fail to load, preventing the user from being able to reach the software update menu in search of a fix.

I'd like to suggest that some kind of button combination on power-up should force systemd to start directly into the software update tool rather than the configured (if any) GUI, this would allow the user to possibly recover their camera in such a situation.

sanjpm, 2020-03-12:
This is a good suggestion now that the software update tool exists. 

I think detecting a combination of both the record button and the encoder wheel being pressed down at boot would be a good way to enter the recovery mode.

oskirby, 2020-03-13:
So, I think the approach for this issue is to have the bootloader check for the record and encoder buttons at boot, then update the kernel command line to boot into systemd's rescue mode. In this mode, the graphical targets are not started, which will not attempt to launch any of the GUIs or network services, but will instead launch anything that declares a WantedBy dependency on rescue.target.

oskirby, 2020-03-13:
We're almost there, except that the PMIC watchdog will ruin the party by rebooting the camera when it comes up in rescue mode since cam-pcUtil does not start in rescue mode... So, more fiddling is necessary.