krontech/chronos-cam-app #165: Bright flashes on 320x32 resolution with max fps after reboot

skronstein, 2020-03-16:
At 320x32 resolution with max fps, some frames are excessively bright. This only happened at max fps, and only after a reboot.
It went away after switching to another resolution and back to 320x32 resolution with max fps.
It also did not occur with fps lowered to 20k, even after a reboot.

tested on 0.4.0-beta27, FPGA revision 3.24

oskirby, 2020-03-26:
These "bright flashes" can be recorded and they will show up in video memory like normal frames. However, when you observe them they are actually just random data, which tends to make them brighter than the valid frames thanks to FPN correction.

The frequency at which they occur seems to be related to the recording speed of the camera, and there appears to be something stateful about this corruption. Switching framerate and back will sometimes make the camera come good, after which no corruption will occur. When examining the register maps of the FPGA and camera both in the good and corrupted states shows identical register mappings.

I suspect that this is more likely due to some kind of race condition in the FPGA, either in the recording sequencer or the deserializer that is getting the write addresses wrong, and it is simply unable to handle these extreme framerates without triggering the bug.

oskirby, 2020-03-26:
This issue also occurs on the LUX2100 if you modify the lux2100 driver to use a smaller minimum vRes.